greiman/DigitalIO #5: Library manager support.

Chris--A, 2016-09-25:
Hi Bill,

As your repo layout is very specific, and seems tailored for doxygen and such, would you be happy with me forking this, modifying the directory structure, and adding it to the Arduino IDE library manager?

I know we are free to fork it, but I have seen authors get cranky at others when they have tried to add their fork to the lib manager.

Great lib by the way, I have been using it for years.

greiman, 2016-09-25:
I will be doing an update soon and it will support the library manger.  

Doxygen is not a problem so I plan to keep Doxygen documentation with the library and use the library manager.

I don't blame authors for getting cranky about your forks.  You don't have your email in box filled with questions about forks that are out of date.  Do a Google search of GitHub to see how many authors suffer from from this problem.

There are now 30 forks of my SdFat library.  Today I have eleven emails about SdFat and most are about out of date forks.

Pull requests are also a pain.  People don't test on all the supported platforms so over 90% of the time there are bugs.

closed 2016-09-25 by greiman

greiman, 2016-09-25:
Chris,

I have been putting off using the Library Manager so I will start with this library.

It may take another try or two to make it acceptable to the Arduino Library Manager.

Chris--A, 2016-09-26:
> I don't blame authors for getting cranky about your forks.

Not mine, but as Authors of code, we are the only ones to blame when we put stuff out publicly. I guess the only thing you can do is not share your code.

There was someone who had quite a tantrum when another user tried to get their fork attached to the library manager.

But thanks for modifying DigitalIO.