greiman/DigitalIO #7: QUESTION on using SoftSPI

mjs513, 2017-09-25:
In looking through the code I noticed you are not setting a CS pin anyway.  Am I assuming it is in effect always selected for that group of pins?  Also if there is no CS function then am I assuming correctly you can only have I device attached for the set of pins selected?

Thanks
Mike

greiman, 2017-09-25:
You must handle CS just like hardware SPI.  You can have multiple devices with different CS pins.  They must all use the same SPI mode.

You only need transfer().  It works like hardware transfer(). 

```
  spi.send(data);
  // same as this transfer call but maybe a bit faster.
  spi.transfer(data);
```

```
  data = spi.receive();
  // same as this transfer call but maybe a bit faster.
  data = spi.transfer(0xff);
```

mjs513, 2017-09-25:
Hi.  Thanks for getting back to me.  Did what you said this morning with the transfers just like normal SPI and it worked.  Got confused when looking at the example (and working late at night).  Now I am trying to get it work from within a 9250 library.

I declared the SoftSPI as spi in the main sketch but now getting spi not defined errors from within the library.  How do I get it work?  Any ideas.  

Thanks
Mike

greiman, 2017-09-25:
Modifying a library to work with SoftSPI may not be easy.  I can't guess about the structure of your 9250 library.

I really don't have time to help you convert a library from hardware SPI to SoftSPI.

A first step is to add an extern statement to the library.

If spi is define in your sketch like this:

```
#include "DigitalIO.h"
SoftSPI<7, 8, 9, 0> spi;
```

You will need something like this to define spi in the library.

```
#include "DigitalIO.h"
extern SoftSPI<7, 8, 9, 0> spi;
```
Good luck, sorry I don't have time to help more.

mjs513, 2017-09-25:
Bill appreciate any help.  Really wasn't looking for you to convert the library, I got that one.  Just didn't know how to do the referencing.   Think I can go from there.  

Thanks again
Mike

closed 2017-09-25 by mjs513