greiman/DigitalIO #8: PlatformIO build fails on espressif8266 platform

ristomatti, 2017-12-06:
Build for Wemos D1 Mini (ESP8266) fails with the below output.

```
> Executing task: platformio run <

[Wed Dec  6 16:50:24 2017] Processing d1_mini (monitor_baud: 115200; platform: espressif8266; board: d1_mini; framework: arduino)
----------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 57 compatible libraries
Looking for dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <DigitalIO> v1.0.0
|-- <OneButton>
Compiling .pioenvs/d1_mini/src/digitalio-test.o
In file included from /home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalIO.h:31:0,
from src/digitalio-test.cpp:2:
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:278:25: error: variable or field 'fastPinMode' declared void
inline void fastPinMode(pin, mode) {
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:278:25: error: 'pin' was not declared in this scope
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:278:30: error: 'mode' was not declared in this scope
inline void fastPinMode(pin, mode) {
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h: In member function 'void DigitalPin<PinNumber>::config(uint8_t, bool)':
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:290:25: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]
{fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:328:5: note: in expansion of macro 'fastPinConfig'
fastPinConfig(PinNumber, mode, level);
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:290:25: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
{fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:328:5: note: in expansion of macro 'fastPinConfig'
fastPinConfig(PinNumber, mode, level);
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h: In member function 'void DigitalPin<PinNumber>::mode(uint8_t)':
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:352:32: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]
fastPinMode(PinNumber, mode);
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/SoftSPI.h: In member function 'void SoftSPI<MisoPin, MosiPin, SckPin, Mode>::begin()':
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:290:25: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]
{fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/SoftSPI.h:55:5: note: in expansion of macro 'fastPinConfig'
fastPinConfig(MisoPin, MISO_MODE, MISO_LEVEL);
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:290:25: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]
{fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/SoftSPI.h:56:5: note: in expansion of macro 'fastPinConfig'
fastPinConfig(MosiPin, MOSI_MODE, !MODE_CPHA(Mode));
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:290:25: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]
{fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/SoftSPI.h:57:5: note: in expansion of macro 'fastPinConfig'
fastPinConfig(SckPin, SCK_MODE, MODE_CPOL(Mode));
^
*** [.pioenvs/d1_mini/src/digitalio-test.o] Error 1
============================================= [ERROR] Took 0.93 seconds =============================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

> Executing task: platformio run <

[Wed Dec  6 16:50:30 2017] Processing d1_mini (monitor_baud: 115200; platform: espressif8266; board: d1_mini; framework: arduino)
----------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 57 compatible libraries
Looking for dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <DigitalIO> v1.0.0
|-- <OneButton>
Compiling .pioenvs/d1_mini/src/digitalio-test.o
In file included from /home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalIO.h:31:0,
from src/digitalio-test.cpp:2:
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:278:25: error: variable or field 'fastPinMode' declared void
inline void fastPinMode(pin, mode) {
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:278:25: error: 'pin' was not declared in this scope
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:278:30: error: 'mode' was not declared in this scope
inline void fastPinMode(pin, mode) {
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h: In member function 'void DigitalPin<PinNumber>::config(uint8_t, bool)':
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:290:25: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]
{fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:328:5: note: in expansion of macro 'fastPinConfig'
fastPinConfig(PinNumber, mode, level);
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:290:25: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
{fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:328:5: note: in expansion of macro 'fastPinConfig'
fastPinConfig(PinNumber, mode, level);
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h: In member function 'void DigitalPin<PinNumber>::mode(uint8_t)':
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:352:32: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]
fastPinMode(PinNumber, mode);
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/SoftSPI.h: In member function 'void SoftSPI<MisoPin, MosiPin, SckPin, Mode>::begin()':
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:290:25: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]
{fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/SoftSPI.h:55:5: note: in expansion of macro 'fastPinConfig'
fastPinConfig(MisoPin, MISO_MODE, MISO_LEVEL);
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:290:25: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]
{fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/SoftSPI.h:56:5: note: in expansion of macro 'fastPinConfig'
fastPinConfig(MosiPin, MOSI_MODE, !MODE_CPHA(Mode));
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/DigitalPin.h:290:25: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]
{fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
^
/home/ristomatti/.platformio/lib/DigitalIO_ID200/src/SoftSPI.h:57:5: note: in expansion of macro 'fastPinConfig'
fastPinConfig(SckPin, SCK_MODE, MODE_CPOL(Mode));
^
*** [.pioenvs/d1_mini/src/digitalio-test.o] Error 1
```

greiman, 2017-12-06:
DigitalIO is not supported on ESP8266.

closed 2017-12-06 by greiman

ristomatti, 2017-12-07:
@greiman ok sorry for reporting this then. You might want to update the `library.properties` file to not include all architectures as DigitalIO is currently listed to support ESP8266.

[image]

greiman, 2017-12-07:
Looks like I did add ESP8266.

I only have an Adafruit HUZZAH ESP8266 but the examples I try seem to compile with Arduino 1.8.4 and esp8266 version 2.3.0.