greiman/DigitalIO #11: Arduino - DigitalIO Library conflicts with SD Library

rjj100, 2019-12-24:
Apparently some functions and constants have the same name in digitalIO and in the SD libraries (as of 12/2019).

The functions having problems are:
badPinNumber(
fastDigitalRead(
fastDigitalWrite(
Please rename in the DigitalIO library.

Also SDA_PIN and SCL_PIN are used in the examples in DigitalIO, and are defined as constants in the SD library.  
Please rename in the DigitalIO library.

I will test the above changes.

rjj100, 2019-12-24:
I made the above changes to DigitalIO library and my sketch compiles without error.

changes:
Roberts-MacBook-Pro:src rjj100$ pwd
/Users/rjj100/Documents/Arduino/libraries/DigitalIO/src
Roberts-MacBook-Pro:src rjj100$ grep -r 'badPinNumber_D(' *
DigitalPin.h:void badPinNumber_D(void)
DigitalPin.h:     badPinNumber_D();
Roberts-MacBook-Pro:src rjj100$ grep -r 'fastDigitalRead_D(' *
DigitalPin.h:bool fastDigitalRead_D(uint8_t pin) {
DigitalPin.h:bool fastDigitalRead_D(uint8_t pin) {
DigitalPin.h:bool fastDigitalRead_D(uint8_t pin) {
DigitalPin.h:bool fastDigitalRead_D(uint8_t pin) {
DigitalPin.h:inline bool fastDigitalRead_D(uint8_t pin) {
DigitalPin.h:  fastDigitalWrite_D(pin, !fastDigitalRead_D(pin));
DigitalPin.h:    return fastDigitalRead_D(PinNumber);
SoftI2cMaster.h:    if (!fastDigitalRead_D(sdaPin)) {
SoftI2cMaster.h:    bool rtn = fastDigitalRead_D(sdaPin);
SoftI2cMaster.h:    if (fastDigitalRead_D(sdaPin)) *data |= 1 << bit;
SoftSPI.h:    if (fastDigitalRead_D(MisoPin)) *data |= 1 << bit;
SoftSPI.h:    if (fastDigitalRead_D(MisoPin)) *rxData |= 1 << bit;
Roberts-MacBook-Pro:src rjj100$ grep -r 'fastDigitalWrite_D(' *
DigitalPin.h:void fastDigitalWrite_D(uint8_t pin, bool level) {
DigitalPin.h:    fastDigitalWrite_D(pin, mode == INPUT_PULLUP);
DigitalPin.h:void fastDigitalWrite_D(uint8_t pin, bool value) {
DigitalPin.h:void fastDigitalWrite_D(uint8_t pin, bool value) {
DigitalPin.h:void fastDigitalWrite_D(uint8_t pin, uint8_t val) {
DigitalPin.h:inline void fastDigitalWrite_D(uint8_t pin, bool value) {
DigitalPin.h:  fastDigitalWrite_D(pin, !fastDigitalRead_D(pin));
DigitalPin.h:  {fastPinMode(pin, mode); fastDigitalWrite_D(pin, level);}
DigitalPin.h:    fastDigitalWrite_D(PinNumber, value);

rjj100, 2019-12-24:
I'll try to attach the modified files.
/Users/rjj100/Documents/Arduino/libraries/DigitalIO/src/DigitalPin.h
/Users/rjj100/Documents/Arduino/libraries/DigitalIO/src/SoftSPI.h
/Users/rjj100/Documents/Arduino/libraries/DigitalIO/src/SoftI2cMaster.h

greiman, 2019-12-24:
They are exactly the same files.  I will include the the entire library in SdFat V2 so you can just use the functions.

closed 2019-12-24 by greiman

greiman, 2019-12-24:
I am not sure how you get a conflict.

I can include both SdFat.h and DigitalIO.h in a sketch with no problem.

```
#include "SdFat.h"
#include "DigitalIO.h"
SdFat sd;
PinIO pin(5);
void setup() {
  sd.begin(10);
  pin.high();
}
void loop() {
}
```

```
Sketch uses 3240 bytes (10%) of program storage space. Maximum is 32256 bytes.
Global variables use 608 bytes (29%) of dynamic memory, leaving 1440 bytes for
local variables. Maximum is 2048 bytes.
```

rjj100, 2019-12-25:
Thank you, you have created some wonderful libraries that I am using in my sketch. 

My sketch is quite large and include lines such as:
if (!SD.begin(4)) {
and
webFile = SD.open(req_target);

So my includes were different. (I am using the SD and DigitalIO libraries from Managed Libraries.)

I  installed and then tried SdFat.h rather than SD.h in my large sketch as you did, and got the error "SD was not declared in this scope."

What would you suggest I do differently, if not modify the DigitalIO library?

rjj100, 2019-12-25:
I did find this, will give it a try in a couple days.

[  https://forum.arduino.cc/index.php?topic=274784.0   ](url)

greiman, 2019-12-25:
I wrote the  version of SdFat used in SD.h over ten years old. It was a beta and I stopped supporting it in 2010 but the Arduino group has not updated it.

I can't fix the conflict with SD.h since the Arduino group maintains the old version of SdFat used with SD.h.

I will soon release a major upgrade of SdFat.  The beta is [here](https://github.com/greiman/SdFat-beta).