greiman/DigitalIO #14: Please Add Support to Atmega8 & Atmega8A

DevDavisNunez, 2022-07-10:
Any chance to add ATMEGA8 AND ATMEGA8 AVR support?

greiman, 2022-07-11:
I don't have these chips or a board to test but I suspect you can add ATmega8 and ATmega8A by editing [GpioPinMap.h](https://github.com/greiman/DigitalIO/blob/master/src/boards/GpioPinMap.h).

change this section:
```
#if defined(__AVR_ATmega168__)\
||defined(__AVR_ATmega168P__)\
||defined(__AVR_ATmega328P__)
// 168 and 328 Arduinos
#include "UnoGpioPinMap.h"
```

To:
```
#if defined(__AVR_ATmega8__)\
||defined(__AVR_ATmega8A__)\
||defined(__AVR_ATmega88__)\
||defined(__AVR_ATmega88A__)\
||defined(__AVR_ATmega88P__)\
||defined(__AVR_ATmega88PA__)\
||defined(__AVR_ATmega168__)\
||defined(__AVR_ATmega168P__)\
||defined(__AVR_ATmega328P__)
// 8 88 168 and 328 Arduinos
#include "UnoGpioPinMap.h"
```

Let me know if it works.

DevDavisNunez, 2022-07-11:
Yes, that is work fine, thanks dear!