greiman/DigitalIO #15: compile failure for rp2040

tbillion, 2022-07-21:
```
Arduino: 1.8.19 (Linux), Board: "Raspberry Pi Pico, 2MB (Sketch: 1536KB, FS: 512KB), 133 MHz, Fast (-Ofast) (maybe slower), Disabled, Disabled, Disabled, Disabled, None, Pico SDK"

In file included from /home/t/Arduino/libraries/DigitalIO/src/DigitalIO.h:31,
                 from /home/t/Arduino/RoboRemotev3v1-tt/RoboRemotev3v1-tt.ino:5:
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:278:13: error: variable or field 'fastPinMode' declared void
  278 | inline void fastPinMode(pin, mode) {
      |             ^~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:278:25: error: 'pin' was not declared in this scope; did you mean 'min'?
  278 | inline void fastPinMode(pin, mode) {
      |                         ^~~
      |                         min
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:278:30: error: 'mode' was not declared in this scope; did you mean 'modf'?
  278 | inline void fastPinMode(pin, mode) {
      |                              ^~~~
      |                              modf
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h: In member function 'void DigitalPin<PinNumber>::config(uint8_t, bool)':
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:290:4: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]
  290 |   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
      |    ^~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:328:5: note: in expansion of macro 'fastPinConfig'
  328 |     fastPinConfig(PinNumber, mode, level);
      |     ^~~~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:290:4: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
  290 |   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
      |    ^~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:328:5: note: in expansion of macro 'fastPinConfig'
  328 |     fastPinConfig(PinNumber, mode, level);
      |     ^~~~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h: In member function 'void DigitalPin<PinNumber>::mode(uint8_t)':
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:352:5: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]
  352 |     fastPinMode(PinNumber, mode);
      |     ^~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/SoftSPI.h: In member function 'void SoftSPI<MisoPin, MosiPin, SckPin, Mode>::begin()':
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:290:4: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]
  290 |   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
      |    ^~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/SoftSPI.h:55:5: note: in expansion of macro 'fastPinConfig'
   55 |     fastPinConfig(MisoPin, MISO_MODE, MISO_LEVEL);
      |     ^~~~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:290:4: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]
  290 |   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
      |    ^~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/SoftSPI.h:56:5: note: in expansion of macro 'fastPinConfig'
   56 |     fastPinConfig(MosiPin, MOSI_MODE, !MODE_CPHA(Mode));
      |     ^~~~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:290:4: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]
  290 |   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
      |    ^~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/SoftSPI.h:57:5: note: in expansion of macro 'fastPinConfig'
   57 |     fastPinConfig(SckPin, SCK_MODE, MODE_CPOL(Mode));
      |     ^~~~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h: In instantiation of 'void DigitalPin<PinNumber>::config(uint8_t, bool) [with unsigned char PinNumber = 5; uint8_t = unsigned char]':
/home/t/Arduino/libraries/PsxNewLib/src/PsxControllerBitBang.h:82:14:   required from 'boolean PsxControllerBitBang<PIN_ATT, PIN_CMD, PIN_DAT, PIN_CLK>::begin() [with unsigned char PIN_ATT = 5; unsigned char PIN_CMD = 4; unsigned char PIN_DAT = 3; unsigned char PIN_CLK = 2; boolean = bool]'
/home/t/Arduino/RoboRemotev3v1-tt/RoboRemotev3v1-tt.ino:251:21:   required from here
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:290:15: error: 'fastPinMode' was not declared in this scope; did you mean 'PinMode'?
  290 |   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
      |    ~~~~~~~~~~~^~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:328:5: note: in expansion of macro 'fastPinConfig'
  328 |     fastPinConfig(PinNumber, mode, level);
      |     ^~~~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h: In instantiation of 'void DigitalPin<PinNumber>::config(uint8_t, bool) [with unsigned char PinNumber = 4; uint8_t = unsigned char]':
/home/t/Arduino/libraries/PsxNewLib/src/PsxControllerBitBang.h:83:14:   required from 'boolean PsxControllerBitBang<PIN_ATT, PIN_CMD, PIN_DAT, PIN_CLK>::begin() [with unsigned char PIN_ATT = 5; unsigned char PIN_CMD = 4; unsigned char PIN_DAT = 3; unsigned char PIN_CLK = 2; boolean = bool]'
/home/t/Arduino/RoboRemotev3v1-tt/RoboRemotev3v1-tt.ino:251:21:   required from here
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:290:15: error: 'fastPinMode' was not declared in this scope; did you mean 'PinMode'?
  290 |   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
      |    ~~~~~~~~~~~^~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:328:5: note: in expansion of macro 'fastPinConfig'
  328 |     fastPinConfig(PinNumber, mode, level);
      |     ^~~~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h: In instantiation of 'void DigitalPin<PinNumber>::config(uint8_t, bool) [with unsigned char PinNumber = 2; uint8_t = unsigned char]':
/home/t/Arduino/libraries/PsxNewLib/src/PsxControllerBitBang.h:84:14:   required from 'boolean PsxControllerBitBang<PIN_ATT, PIN_CMD, PIN_DAT, PIN_CLK>::begin() [with unsigned char PIN_ATT = 5; unsigned char PIN_CMD = 4; unsigned char PIN_DAT = 3; unsigned char PIN_CLK = 2; boolean = bool]'
/home/t/Arduino/RoboRemotev3v1-tt/RoboRemotev3v1-tt.ino:251:21:   required from here
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:290:15: error: 'fastPinMode' was not declared in this scope; did you mean 'PinMode'?
  290 |   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
      |    ~~~~~~~~~~~^~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:328:5: note: in expansion of macro 'fastPinConfig'
  328 |     fastPinConfig(PinNumber, mode, level);
      |     ^~~~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h: In instantiation of 'void DigitalPin<PinNumber>::config(uint8_t, bool) [with unsigned char PinNumber = 3; uint8_t = unsigned char]':
/home/t/Arduino/libraries/PsxNewLib/src/PsxControllerBitBang.h:85:14:   required from 'boolean PsxControllerBitBang<PIN_ATT, PIN_CMD, PIN_DAT, PIN_CLK>::begin() [with unsigned char PIN_ATT = 5; unsigned char PIN_CMD = 4; unsigned char PIN_DAT = 3; unsigned char PIN_CLK = 2; boolean = bool]'
/home/t/Arduino/RoboRemotev3v1-tt/RoboRemotev3v1-tt.ino:251:21:   required from here
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:290:15: error: 'fastPinMode' was not declared in this scope; did you mean 'PinMode'?
  290 |   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}
      |    ~~~~~~~~~~~^~~~~~~~~~~
/home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:328:5: note: in expansion of macro 'fastPinConfig'
  328 |     fastPinConfig(PinNumber, mode, level);
      |     ^~~~~~~~~~~~~
exit status 1
Error compiling for board Raspberry Pi Pico.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

will not compile for any rp2040 and unfortunately i need this to work as it is a dependency for the library i need,

greiman, 2022-07-22:
I have no idea what is going on. Your compile errors don't match the library source.  The uint8_t types are missing.

Error:

> /home/t/Arduino/libraries/DigitalIO/src/DigitalPin.h:278:25: error: 'pin' was not declared in this scope; did you mean 'min'?
>   278 | inline void fastPinMode(pin, mode) {
>       |                         ^~~        ^~~~~~~~~~~
> 

Source from  DigitalPin.h:278:
```
inline void fastPinMode(uint8_t pin, uint8_t mode) {
  pinMode(pin, mode);
}
```

Try compiling the testFastDigital example.  It compiles for my rp2040.

11chri, 2022-09-27:
I just had the same issue on rp2040; I used platformio to import the library.

The fix for fastPinMode was introduced after latest tagged version 1.0.0 and is an untagged release (latest master).
[image]

@greiman Could you tag the latest commit in master with a new version 1.0.1, please? Then the issue with platformio should also be gone as I suppose it will just check out latest tagged release.

With latest commit in master, it also compiles for my rp2040

11chri, 2022-09-28:
I think, this topic can be closed as there is a new released version 1.0.1 available with the required bugfix.
I also just noticed that platformio now references to the latest release 1.0.1, https://registry.platformio.org/libraries/greiman/DigitalIO/versions

closed 2022-10-04 by tbillion